leoherzog/WundergroundStationForwarder #3: Failed Execution

moulari, 2020-12-11:
Hello,

First of all great project been looking for something like this for quite some time. 

I can't seem to get the forwarding to work

for Weathercloud I get

_TypeError: (intermediate value).inTomm is not a function
    at updateWeathercloud_(forwarder_pws_kattau2:119:111)_

for PWSWeather  I get

_Exception: Invalid Argument: http://www.pwsweather.com/pwsupdate/pwsupdate.php?ID=PWSKATTAU&PASSWORD=xxxxxxxxxxxx&dateutc=2020-12-11%2019:57:18&tempf=35.7&windspeedmph=0&winddir=179&baromin=29.44&dewptf=34.7&humidity=96&uv=0
    at updatePWSWeather_(forwarder_pws_kattau2:97:30)_ 

Note: Password is correct, just changed it for the copy and paste!

Thanks!

leoherzog, 2020-12-13:
I think you just found a typo! Sorry about that. Change the second-to-last line from:

`Number.prototype.inTohmm = function() { return this * 25.4; }`
to
`Number.prototype.inTomm = function() { return this * 25.4; }`

And see if that works?

moulari, 2020-12-14:
Thanks, works flawlessly now. 

PWSWeather, ist not reporting, 

I get

Exception: Invalid argument: http://www.pwsweather.com/pwsupdate/pwsupdate.php?ID=PWSKATTAU&PASSWORD=xxxxxxxxxxxxxxx&dateutc=2020-12-14%2010:16:18&tempf=41.3&windspeedmph=0&winddir=122&baromin=29.94&dewptf=39.7&humidity=94&uv=0
    at updatePWSWeather_(forwarder_pws_kattau2:97:30)

If i copy that string: http://www.pwsweather.com/pwsupdate/pwsupdate.php?ID=PWSKATTAU&PASSWORD=xxxxxxxxxxxxxxx&dateutc=2020-12-14%2010:16:18&tempf=41.3&windspeedmph=0&winddir=122&baromin=29.94&dewptf=39.7&humidity=94&uv=0 and paste it into the brwoser address bar i get a page with the message: _Data Logged and posted in METAR mirror._

_Note: Password is correct, just changed it for the copy and paste!_

leoherzog, 2020-12-14:
Hrm. If you change the UTC date (around line 88) in the `updatePWSWeather_()` function from:
```
request += '&dateutc=' + Utilities.formatDate(new Date(station.obsTimeUtc), 'UTC', 'yyyy-MM-dd HH:mm:ss');
```
to
```
request += '&date=' + Utilities.formatDate(new Date(station.obsTimeUtc), 'UTC', 'yyyyMMdd');
request += '&time=' + Utilities.formatDate(new Date(station.obsTimeUtc), 'UTC', 'HHmm');
```
does it work?

leoherzog, 2020-12-14:
Hrm. I was able to get it working with all three parameters. If you have these three lines, does it work for you?

```
request += '&dateutc=' + Utilities.formatDate(new Date(station.obsTimeUtc), 'UTC', 'yyyy-MM-dd HH:mm:ss');
request += '&date=' + Utilities.formatDate(new Date(station.obsTimeUtc), 'UTC', 'yyyyMMdd');
request += '&time=' + Utilities.formatDate(new Date(station.obsTimeUtc), 'UTC', 'HHmm');
```

moulari, 2020-12-15:
Thanks for the support

hm.. i tried the changes to the PWSWeather Function, but no change. I even created a new station in PWSWeather but no change. Still the same issue

```
Exception: Invalid argument: http://www.pwsweather.com/pwsupdate/pwsupdate.php?ID=PWSKATTAU&PASSWORD=xxxxxxxxxxxxxxx&dateutc=2020-12-14%2010:16:18&tempf=41.3&windspeedmph=0&winddir=122&baromin=29.94&dewptf=39.7&humidity=94&uv=0
    at updatePWSWeather_(forwarder_pws_kattau2:97:30)
```
If I just post the URL into the browser I get the same message as mentioned before and if I do that manually every 5 minutes the station in PWSWeather reports as active. Since the URL works for the reporting it seems not to be a problem with the parameters rather with the posting of the URL.

leoherzog, 2020-12-15:
Okay. I don't have a weather station of my own with all of these features, so I can't test it myself. What about if you change
```
if (station.uv != null) request += '&uv=' + station.uv;
```
to
```
if (station.uv != null) request += '&UV=' + station.uv;
```
in the PWSWeather? You can remove the `&date=` and `&time=` lines.

leoherzog, 2020-12-18:
I did some testing, I think this change might be the ticket. When you get a minute, @moulari, give 'er a shot and let me know if that fixes it for you!

leoherzog, 2021-04-23:
I believe this is working now.

closed 2021-04-23 by leoherzog