leoherzog/WundergroundStationForwarder #4: WindGust is missing for PWSWeather

judgetom, 2021-04-23:
Hello,

in the function updatePWSWeather_() the following line is missing:
  _if (station.imperial.windGust != null) request += '&windgustmph=' + station.imperial.windGust;_

By adding this line the windGust will also be provided.

Best regards, and thanks for the script,
judgetom

leoherzog, 2021-04-23:
Thanks! I was basing this built URL off of the PWSWeather uploader in `acuparse`.

https://gitlab.com/acuparse/acuparse/-/blob/dev/src/fcn/cron/uploaders/pwsweather.php

Is this parameter actually supported by PWSWeather? They don't have any API docs :(

leoherzog, 2021-04-23:
Seems to be referenced by other projects!

https://github.com/OurColonial/WeatherLink-to-PWSweather

https://pywws.readthedocs.io/en/latest/api/pywws.service.pwsweather.html

judgetom, 2021-04-23:
Yes, it works well since I added it.

leoherzog, 2021-04-23:
Thanks again!

commit 1d12aa78 (2021-04-23): Add PWSWeather Wind Gust

Fixes #4. Thanks @judgetom!
closed 2021-04-23 by leoherzog