leoherzog/WundergroundStationForwarder #6: no precipitation/UV export to PWS weather

Dainiux, 2021-11-11:
UV export to PWS Weather does not work. 
No precipitation data export to PWS Weather

leoherzog, 2021-11-11:
Thanks for the heads up! I don't have a weather station with a UV sensor so I can't test, but can you try changing
`if (station.uv != null) request += '&uv=' + station.uv;`
to
`if (station.uv != null) request += '&UV=' + station.uv;`
on [Line 101](https://github.com/leoherzog/WundergroundStationForwarder/blob/master/code.gs#L101) and see if that fixes it?

Dainiux, 2021-11-14:
Hello
problem was with Solar Radiation, not UV. But I have checked a code, it is
missing precipitation and solar radiation export to PWS at all. I wrote it,
you can change pws export code into this one:

function updatePWSWeather_() {

  let station = JSON.parse(CacheService.getScriptCache().get(
'weatherstation'));

  let request = 'https://pwsupdate.pwsweather.com/api/v1/submitwx';
  request += '?ID=' + pwsWeatherStationID;
  request += '&PASSWORD=' + pwsWeatherPassword;
  request += '&dateutc=' + Utilities.formatDate(new Date(station.obsTimeUtc
), 'UTC', 'yyyy-MM-dd HH:mm:ss');
  request += '&tempf=' + station.imperial.temp;
  if (station.imperial.windSpeed != null) request += '&windspeedmph=' +
station.imperial.windSpeed;
  if (station.imperial.windGust != null) request += '&windgustmph=' +
station.imperial.windGust;
  if (station.winddir != null) request += '&winddir=' + station.winddir;
  if (station.imperial.pressure != null) request += '&baromin=' + station.
imperial.pressure;
  if (station.imperial.dewpt != null) request += '&dewptf=' + station.
imperial.dewpt;
  if (station.humidity != null) request += '&humidity=' + station.humidity;
  if (station.solarRadiation != null) request += '&solarradiation=' +
station.solarRadiation;
  if (station.uv != null) request += '&uv=' + station.uv;
  if (station.imperial.precipRate != null) request += '&rainin=' + station.
imperial.precipRate;
  if (station.imperial.precipTotal != null) request += '&dailyrainin=' +
station.imperial.precipTotal;
    request +=
'&softwaretype=appsscriptwundergroundforwarder&action=updateraw';

  let response = UrlFetchApp.fetch(request).getContentText();

  console.log(response);

  return response;

}

closed 2021-11-15 by leoherzog

leoherzog, 2021-11-15:
Helpful! Thank you!

commit ca962eeb (2021-11-15): Added PWSWeather Solar Radiation and Precip Rate
Closes #6