leoherzog/WundergroundStationForwarder #9: Scheduling Script

eyousey, 2022-09-13:
The code does not seem to be setting up automatic scheduling.  Trying to link WU to PWSWeather.com.

Execution status only shows when I run it manual.

leoherzog, 2022-09-25:
What do you see in the `⏰` Triggers section on the left sidebar? As long as you have `const updatePWSWeather = true;` set on [`Line 37`](https://github.com/leoherzog/WundergroundStationForwarder/blob/main/code.gs#L37) when you run `Schedule()`, it should start running that automatically.

eyousey, 2022-09-26:
So, nothing showed up in the triggers section.  I did have the
updatepwsweather set to true.  I just put in a manual trigger, but my
pwsweather station is showing as "inactive".  Even when first testing it,
it was showing as inactive.

I manually set a trigger to run every 5 minutes, like in your code, and it
seems to be running, but it is not either getting info from WU to send to
pwsweather or it is not sending data to pwsweather.  I'm not a programmer,
but I can see by your code comments that pwsweather doesn't really have an
api.

Let me know if you have any ideas, and I can try them.

Thanks.

leoherzog, 2022-09-27:
Does anything print to the console (the "log" on the bottom of the screen) when you press ▷Run to run the `Schedule` function?

eyousey, 2022-09-27:
Yes...

```
Execution log
7:53:33 AM	Notice	Execution started
7:53:33 AM	Notice	Execution completed
```

leoherzog, 2022-09-27:
Could you take a screenshot or copy-paste lines 10 through 90 in your project as they exist now?

eyousey, 2022-09-27:
Here are lines 10 thru 90.  Edited out my station ID and keys.
```
const datasource = 'ibm'; // 'ibm' (wunderground), 'acurite' (myacurite), 'davis' (weatherlink), or 'weatherflow' (tempestwx)

const ibmAPIKey = 'key';
const ibmStationId = 'id';
// or
const acuriteUsername = 'xxxxxx@example.com';
const acuritePassword = 'xxxxxxxxxxxxxxxxxx';
const acuriteHubName = 'xxxxxxxxxxxxxxxx';
const acuriteStationName = 'xxxxxxxxxxxxxxxx';
// or
const davisApiKey = 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx';
const davisApiSecret = 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx';
const davisStationName = 'xxxxxxxxxxxxxxxx';
// or
const weatherflowPUT = 'xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx';
const weatherflowStationId = 'xxxxx';

// Sending data

const updateWunderground = false;
const wundergroundStationId = 'KXXXXXXXXXX';
const wundergroundStationKey = 'xxxxxxxx';
///
const updateWindy = false;
const windyAPIKey = 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx';
const windyStationId = '0';
///
const updatePWSWeather = true;
const pwsWeatherAPIKey = 'key';
const pwsWeatherStationID = 'id';
///
const updateWeatherCloud = false;
const weathercloudAPIKey = 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx';
const weathercloudID = 'xxxxxxxxxxxxxxxx';
const hasWeatherCloudPro = false;
///
const updateOpenWeatherMap = false;
const openWeatherMapAPIKey = 'xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx';
const openWeatherMapStationID = 'xxxxxxxxxxxxxxxx';

/*
  ____          _   _       _     _____    _ _ _     ____       _               
 |  _ \  ___   | \ | | ___ | |_  | ____|__| (_) |_  | __ )  ___| | _____      __
 | | | |/ _ \  |  \| |/ _ \| __| |  _| / _` | | __| |  _ \ / _ \ |/ _ \ \ /\ / /
 | |_| | (_) | | |\  | (_) | |_  | |__| (_| | | |_  | |_) |  __/ | (_) \ V  V / 
 |____/ \___/  |_| \_|\___/ \__| |_____\__,_|_|\__| |____/ \___|_|\___/ \_/\_/  

*/

let version = 'v2.0.1';

function Schedule() {
  ScriptApp.getProjectTriggers().forEach(trigger => ScriptApp.deleteTrigger(trigger));
  switch (datasource) {
    case 'ibm':
      refreshFromIBM_();
      ScriptApp.newTrigger('refreshFromIBM_').timeBased().everyMinutes(1).create();
      break;
    case 'acurite':
      refreshFromAcurite_();
      ScriptApp.newTrigger('refreshFromAcurite_').timeBased().everyMinutes(1).create();
      break;
    case 'davis':
      refreshFromDavis_();
      ScriptApp.newTrigger('refreshFromDavis_').timeBased().everyMinutes(1).create();
      break;
    case 'weatherflow':
      refreshFromWeatherflow_();
      ScriptApp.newTrigger('refreshFromWeatherflow_').timeBased().everyMinutes(1).create();
      break;
  }
  if (updateWunderground) ScriptApp.newTrigger('updateWunderground_').timeBased().everyMinutes(1).create();
  if (updateWindy) ScriptApp.newTrigger('updateWindy_').timeBased().everyMinutes(5).create();
  if (updatePWSWeather) ScriptApp.newTrigger('updatePWSWeather_').timeBased().everyMinutes(5).create();
  if (updateWeatherCloud) ScriptApp.newTrigger('updateWeatherCloud_').timeBased().everyMinutes(hasWeatherCloudPro ? 1 : 10).create();
  if (updateOpenWeatherMap) ScriptApp.newTrigger('updateOpenWeatherMap_').timeBased().everyMinutes(1).create();
  console.log('Scheduled! Check Executions ☰▶ tab for status.');
  checkGithubReleaseVersion_();
}
```

leoherzog, 2022-09-27:
Hrm. Everything looks correct... I'm not sure why you wouldn't get
```
10:10:14 AM  Notice	Execution started
10:10:16 AM  Info	{"time":1664287810000,"temp":{"f":49.6,"c":9.8},"dewpoint":{"f":45.6,"c":7.6},"windSpeed":{"mph":19,"mps":8},"windGust":{"mph":21,"mps":9},"winddir":293,"pressure":{"inHg":30,"hPa":1015.9},"humidity":86,"precipRate":{"in":0,"mm":0},"precipTotal":{"in":0.01,"mm":0.25}}
10:10:17 AM  Info	Scheduled! Check Executions ☰▶ tab for status.
10:10:18 AM  Notice	Execution completed
```
when you run `Schedule`. When you ran it for the first time, did anything pop up and ask you for authorization? Did you click "allow"?

leoherzog, 2022-09-27:
Sorry for the trouble, but could you create another new project and run through the steps again? The `Schedule` function looks correct in your code but the log that you're describing means that nothing actually happened. Perhaps it's a bug on Google's side with your Apps Script project? I'm fresh out of ideas otherwise.

eyousey, 2022-09-27:
Just made a new scripts project, put in the code, added in my stations and keys, set pwsweather.com to true, saved it, ran it, and pretty much the same result.  It asked for permission, I granted it, and no triggers were created and all I got for out put was...

```
Execution log
3:57:09 PM	Notice	Execution started
3:57:09 PM	Notice	Execution completed
```

I also double checked my keys.

Is there maybe an API you have setup that I'm not pulling in?  Or some other addon?

eyousey, 2022-09-28:
So.. I thinking, maybe the script can't grab the WU data.  If it can't grab it, it can't upload it.

I saw this line of code.
`let ibmConditions = fetchJSON_('https://api.weather.com/v2/pws/observations/current?stationId=' + ibmStationId + '&format=json&units=e&numericPrecision=decimal&apiKey=' + ibmAPIKey);`

Inserted my ID and key into the link above, and got this via a web browser.
`{"metadata":{"transaction_id":"##############"},"success":false,"errors":[{"error":{"code":"CDN-0001","message":"Invalid apiKey."}}]}`

It says the key is invalid, but I'm using the same key to upload info to WU, and has been working for over a year.

eyousey, 2022-09-28:
Ok... more digging/learning.  I was using my station key... read your code some more, and saw I needed an API key.  So I generated one and now the above link gives me ...

`{"observations":[{"stationID":"STATIONID","obsTimeUtc":"2022-09-28T12:30:06Z","obsTimeLocal":"2022-09-28 08:30:06","neighborhood":"CITY","softwareType":null,"country":"US","solarRadiation":null,"lon":-81.531,"realtimeFrequency":null,"epoch":1664368206,"lat":40.89,"uv":null,"winddir":292,"humidity":99.0,"qcStatus":1,"imperial":{"temp":48.7,"heatIndex":48.7,"dewpt":48.5,"windChill":48.7,"windSpeed":1.7,"windGust":1.7,"pressure":30.12,"precipRate":0.00,"precipTotal":0.00,"elev":1171.0}}]}`

But still no scheduling is set up, and not upload to PWSweather.com.

eyousey, 2022-09-28:
Ok.. found the issue... 

`function PWS_Logger2() {
<inserted your code here>
}`

This was the code that google put in when I started the project.  This did not work.  So, I removed the function code that was put in buy google, and only had your code.  This gave me logged output showing the weather stations current info like you showed in an earlier post.

PWSweather.com station log shows the data from my weather stations.  It's working.

Sorry about all the trouble, but I'm no programmer and didn't know I needed to paste only your code and over write the default code that google dropped in there.

closed 2022-09-28 by eyousey

leoherzog, 2022-09-30:
Ohhhh! The function that you were running along the top bar was `function myFunction()` or whatever they put in there by default! Great deduction work. So glad you figured it out, I was out of troubleshooting ideas!

eyousey, 2022-09-30:
Thanks for the creating the code!!  I wanted to get my weather station reporting to PWSWeather.com so I can link it to the irrigation controller I plan to purchase in the spring.  My weather station is compatible with WU and Weather Cloud, but not PWS.  This script has done the trick, and I didn't need to have a computer keeping things up to date.

For your documentation, you might want to add a note about over writing the default code and also making sure you have an API key, not a station key, from WU.  Not sure about the other services if you expressly need an API key.  It also looks like the WU key expires in 6 months.  Hopefully it stays active as long as you're using.  We'll see.

Thanks again.

leoherzog, 2022-09-30:
As soon as I read your message when you closed the issue, I tweaked the instructions :)

https://github.com/leoherzog/WundergroundStationForwarder/commit/5811413f56767d6c45bfa93ce9176ccae672a82e