leoherzog/WundergroundStationForwarder #17: CWOP

aolfish, 2023-08-29:
Line 54 should = true
Humidity not being reported from Acurite to CWOP

leoherzog, 2023-08-29:
Hi @aolfish!

As to your first point, and as [the instructions explain](https://github.com/leoherzog/WundergroundStationForwarder/blob/main/README.md#setup), all of the various forwarding destinations start off as `false`. You're correct that if you want to begin reporting to CWOP, you'll want to set line 54 to true in _your_ project... but not everyone will want to do so. 

As for the humidity thing, it is currently set up to [receive from MyAcurite](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.4.2/code.gs#L311-L312) and [send to CWOP](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.4.2/code.gs#L828). I can confirm that my station is showing up on FindU with the current relative humidity listed along the top. Where/how are you viewing your station's CWOP data?

leoherzog, 2023-08-29:
@aolfish I think I found the problem! Thank you for making me aware of it. Give the latest version a try and see if it fixes the problem for you.

closed 2023-08-29 by leoherzog

aolfish, 2023-08-29:
Now I'm getting the following error

Error	
Problem fetching https://api.weather.com/v2/pws/observations/current?stationId=KXXXXXXXXXX&format=json&units=e&numericPrecision=decimal&apiKey=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
fetchJSON_	@ code.gs:900
refreshFromIBM_	@ code.gs:121
Schedule	@ code.gs:78

leoherzog, 2023-08-29:
Are you trying to fetch your readings from Wunderground or from MyAcurite? Which data source have you chosen on [Line 10](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.4.4/code.gs#L10)?

aolfish, 2023-08-29:
Thank you.  Was in a hurry to get it reset up and missed Line 10.   Humidity is showing up on data received now.

commit fea885bf (2023-08-29): Bugfix for MyAcurite Humidity and Heat Index Calc

Fixes #17