leoherzog/WundergroundStationForwarder #18: Distinguish Between "Rain Over Last 24 Hours" and "Rain Since Midnight"

leoherzog, 2023-09-01:
Update every sender and receiver to distinguish between these two values

aolfish, 2023-09-06:
CWOP has input values showing : (rain 1hr), (rain 24hr) & (rain mn).  My weather reports only showed values under the (rain 1hr) column.

leoherzog, 2023-09-08:
Added support to cwop.rest to distinguish the two.

https://github.com/leoherzog/cwop.rest/commit/06f74498af4cc5819ac5c95ee537845c494d21a8

https://web.archive.org/web/20230716040809/http://www.wxqa.com/faq.html item number 7

closed 2023-09-08 by leoherzog

leoherzog, 2023-09-08:
Done!

aolfish, 2023-09-12:
Finally got a smidge of rain today.  Only 0.06 in.  My CWOP weather reports on findu.com only showing rain in 1hr column.  Nothing (0.00) in the 24hr and MN columns.  Using the V2.5.0

leoherzog, 2023-09-12:
@aolfish What is your data source?

aolfish, 2023-09-12:
Sorry ... Acurite

leoherzog, 2023-09-13:
Sure. If you uncomment [the `console.log(JSON.stringify(acuriteConditions));` line in the `refreshFromMyAcurite_()` function](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.5.0/code.gs#L274), you'll start to see what data I get from them in the ☰▶ Execution Log on the left. There's a "Rainfall" one that I can grab for [the precipitation rate](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.5.0/code.gs#L325-L329), but MyAcurite doesn't give me any stats on hourly rainfall or totals over 24 hours/since midnight. Unless there's another "get historical data" API endpoint that I don't know of, I think we're out of luck. It's tough to develop for this API specifically since it's an undocumented, internal API that could change or go away at any time. Official, Acurite offers no API for MyAcurite.

commit 2e0bca92 (2023-09-08): Distinguish Between Precipitation Rate Intervals

Current Rate, Last Hour, Since Midnight, and Last 24 Hours
Closes #18