leoherzog/WundergroundStationForwarder #19: Issues while forwardig WeatherFlow Tempest data to WU

karss79, 2023-09-16:
Hi,
I have been trying your AppScript code from version 2.3.0 to the current 2.5.0, to upload my Tempest Weather Station data to Weather Underground, unfortunately the WU upload doesn't seem to work properly. I see discrepancies in the pressure data, wind speed, wind gust data, rain rate and also rainfall.
The pressure data fluctuates between two values, wind speed and wind gust seems are reported in steps, for example wind speed is reported in steps of 1.6, and so is the wind gust. Similarly, I think the cumulative rainfall data is also reported in incremental values of 0.25/0.26 mm. When it comes to rain rate, it is not able capture this information.
The rest of the data like, temperature, dew point, wind direction, radiation and UV seem to be okay.
In order to have a comparison of the data forwarded by your appscript to WU and also the data forwarded by Weatherflow to WU, I setup two stations in Weather Underground, but both receiving weather data from the same Tempest Station, while one receives the data from WeatherFlow in 5 minutes interval and the other through your appscript code in 1 minute interval. I have attached a comparison of the two sets of data in the form of a screenshot attachment, to the left is the data from Weatherflow and to the right is the data forwarded by AppScript.
Hope this information helps you to refine your future versions of Wunderground Station Forwarder.
Thanks

[image]

karss79, 2023-09-16:
Another set of data from today!
[image]

leoherzog, 2023-09-21:
This is fascinating! I also use a Tempest as my main station and I've always used my code over the built-in one because of the higher refresh rate. Thanks for collecting this.

For the Pressure, I'm pulling down pressure in hPa from their API, converting to inHg,

https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.5.0/code.gs#L472-L475

and sending it along:

https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.5.0/code.gs#L617

The issue is in the conversion there. I was only converting to inHg up to one decimal point, which was causing the lack of detail in that pressure graph. Possibly fixed in be7e58a585e27dd131c5dc032283054f5550a427.

Similar problem with the Wind Speeds! Looking to fix that in a bit with your suggestion.

https://github.com/leoherzog/WundergroundStationForwarder/commit/be7e58a585e27dd131c5dc032283054f5550a427#commitcomment-127946364

karss79, 2023-09-21:
Posted the below comment this morning as comment under your code update. Posting it here again:

Hi leoherzog,

Thanks for the updated code, it fixes the pressure issue in Wundergroud. Taking cue from your updated code, I changed the FixedNumber values in line number 460 and 466 from 0 to 1, this fixed the issue with wind speed and wind gust reporting in steps of 1.6 km/h in Wunderground.

460 "mph": Number(weatherflowConditions.obs[0].wind_avg).mpsToMPH().toFixedNumber(1),
466 "mph": Number(weatherflowConditions.obs[0].wind_gust).mpsToMPH().toFixedNumber(1),

Now the left out issue is Rain Rate, there is some error in calculation of rain rates as it reports very low value, may be you can look into it. If this issue is fixed then your Script is complete and free of bugs when it comes to reporting WeatherFlow Tempest data to Weather Underground. Attached is a screenshot of the WU display before and after updating the code. The vertical line marks the point of change. Also another screenshot showing the RainRate as reported by WeatherFlow and AppScript Forwarder to Wundergound. This data is from this morning.
Thanks

[image]

[image]

karss79, 2023-09-21:
For your information, my computer knowledge is very basic, don't know anything about python or coding, just tried my luck and in the wind case it worked.

leoherzog, 2023-09-21:
Great, thanks. Increased the precision of wind speed units in https://github.com/leoherzog/WundergroundStationForwarder/commit/8b5fd2c0e718adb81776f1151a19996f66163d39. I'll look into rain rates next.

leoherzog, 2023-09-21:
What WeatherFlow's Tempest API gives me looks like this for precipitation data:

```json
{
  "precip": 0,
  "precip_accum_last_1hr": 0,
  "precip_accum_local_day": 0,
  "precip_accum_local_day_final": 0,
  "precip_accum_local_yesterday": 0,
  "precip_accum_local_yesterday_final": 0,
  "precip_analysis_type_yesterday": 0,
  "precip_minutes_local_day": 0,
  "precip_minutes_local_yesterday": 0,
  "precip_minutes_local_yesterday_final": 0,
}
```
What is `precip`? I assume it's the current rate at this very moment (in mm/s?), but I have no idea. [Their docs](https://weatherflow.github.io/Tempest/api/swagger/#!/observations/getStationObservation) don't say (under Observations → Station). I'll try and reach out to their support to see what they're sending in their built-in WU integration.

karss79, 2023-09-22:
I am not sure what 'precip' means. Since, WeatherFlow averages data to one minute, I guess it could be rain in mm/minute or in/minute. I am giving below the formulae that WeatherFlow uses to calculate RainRate.
Hope this information helps!
Thanks for your efforts.
[image]

karss79, 2023-09-22:
I am testing your script with the latest code updates and today I just had a slight drizzle amounting to 0.15 mm rain accumulation in my Tempest device as reported by the Tempest app and also the app forwarded data to Wunderground. But when the rain data was sent to Wunderground through your AppsScriptForwarder it reported a rain accumulation of 0.25mm against the actual accumulation of 0.15mm. 
So, I edited your script at line no. 506 and 510 and changed the FixedNumber value to 3 from 2, this change fixed the rain accumulation issue. A value of 2 is okay for weather stations using tipping bucket rain gauges, where each tip is measured in increments of 0.01 inch, but the haptic rain sensor in Tempest can record much lower rain accumulation. For example on one day my station had recorded a trace rainfall of 0.03 mm, this kind of very low rain measurement is not possible with tipping bucket rain gauges.
By the way, thanks again for your efforts to solve the issues asap. Definitely your script will be useful to many.

leoherzog, 2023-09-22:
The wonderful people at Weatherflow Support had this to say:

> Hi there, thanks for writing in! "Precip" = rain rate in mm/hr, for that minute. The built-in WU integration sends a single data point (from the 1-minute obs) every five minutes (or at least we try to, sometimes this connection slows down on us but that's unfortunately out of our control) which is why the graph on the left has a more stair-stepped look.

I'm following up with them asking if they have any specific insights as to the green line discrepancy in your Rain Rate graph.

karss79, 2023-09-25:
Hi leoherzog, Had a couple of rainfall events during the past two days and I didn't see any change in the rainrate curve displayed in WU. I think somewhere there is small calculation or conversion error, because the RR value gets reported but the magnitude of the value is several orders small. Hope WeatherFlow support helps you to fix the issue.

karss79, 2023-10-01:
After some experimentation, I managed to fix the rain rate discrepancy in Wunderground. For some reason the 'preciprate' as reported by WeatherFlow is divided by 60 and then reported as rainrate by WU. To correct this issue, I multiplied 'preciprate' by 60 and the issue got fixed.
I have implemented the following changes to your latest code to make the values reported in WU more precise and as close as possible to the one reported by WeatherFlow:
Line No. 461 - FixedNumber 2
Line 467 - FixedNumber 2
Line 503 - FixedNumber 3
Line 507 - FixedNumber 3
Line 511 - FixedNumber 3
Line 503 - * 60
Line 504 - * 60
The above changes works well for Tempest data reporting to Wunderground.
Thanks for your efforts.

leoherzog, 2023-10-02:
I guess Weatherflow is giving us per minute and Wunderground is expecting per hour?

karss79, 2023-10-02:
I don't think there is a conversion issue at the WU end. After the above correction for rainrate, I checked the data forwarded by AppsScriptForwarder obtained from WeatherFlow to WU at the Executions menu in Google Script, here I could see that whatever value that is reported for 'rainrate' is exactly reflected in WU.

leoherzog, 2023-10-02:
I don't think that Miles Per Hour needs to have two sig figs of accuracy (It's 8.73mph windspeed?), nor do inches to three. Do you really need to know rain to 0.001 of an inch? I don't think most sensors are even that accurate. The common values I've seen from most sensors and on most websites are are zero or one sig fig for wind and inches (14.3mph or 1.2" of rain) for meteorological purposes. Would you agree?

karss79, 2023-10-02:
I agree with you if the data is presented in imperial units, but when you do the conversion from imperial to metric, the number of decimal points becomes significant. For example, I have done a simple inch to mm conversion for a number with four, three, two and one decimal points, you can see how it affects the value calculated in mm with just two decimal points.
**Inch | mm**
1.2345 | 31.36
1.234 | 31.34
1.23 | 31.24
1.2 | 30.48
Hope you get what I mean.

leoherzog, 2023-10-02:
I do! Great point. Most weather data destinations support several sig figs, so perhaps I'll bump them all up then.

karss79, 2023-10-02:
More the merrier!