leoherzog/WundergroundStationForwarder #21: no more data on Meteo sites

yo2ldk, 2024-05-31:
Hi,
I have this error:
Problem fetching https://api.weather.com/v2/pws/observations/current?stationId=xxxxx&format=json&units=e&numericPrecision=decimal&apiKey=xxxxxxx

fetchJSON_	@ Code.gs:920
refreshFromIBM_	@ Code.gs:120
Schedule	@ Code.gs:77

something was changed ?

Thank you for all your work !!

leoherzog, 2024-05-31:
Did you Wunderground API Key expire?

https://www.wunderground.com/member/api-keys

yo2ldk, 2024-05-31:
Thank you for reply,  
no, not was expired, I make it anyway a new one
also Meteo station have an upload key for Wunderground and still ok
I will try it more..

yo2ldk, 2024-05-31:
no more errors on Run
maybe it was something with my key,,
Thank you for support and this nice app!

closed 2024-05-31 by yo2ldk