leoherzog/WundergroundStationForwarder #22: forwarding to CWOP to other sites

CT2IRJ, 2024-06-18:
Hi.

This is a great tool for those who like myself owns a WS wich does not provide data forwarding to CWOP.

Is there a chance that provides the opposite, I mean to grab data from CWOP stations and forward them to any other weather service?

I'll explain, i have in cooperation with AMSAT-PO a few remote weather stations that work only via APRS.
Our city hall asked us if we can feed them the data to monitor weather conditions in case of wildfires and such.

Their main goal was to create a dashboard like webpage to display the conditions.
I was thinking on the way to achieve that was to forward the specific weather stations to WU (or other service) and with the API of such service grab the data to their use...

thanks in advance

leoherzog, 2024-06-18:
Great question. It would take some work. Lemme look into that for you to see if it's possible.

CT2IRJ, 2024-06-18:
Hi Leo.

Thanks for considering it.

Salomão Fresco

Callsign: CT2IRJ

GRID Locator: IM59re -- CQ Zone 14 / ITU Zone 37

QTH: Fazendas de Almeirim, Portugal

leoherzog, 2024-06-18:
Added a new datasource option for APRS (CWOP) weather stations in the latest release, [v2.6.0](https://github.com/leoherzog/WundergroundStationForwarder/releases/tag/v2.6.0). This uses the wonderful [aprs.fi API](https://aprs.fi/page/api). You'll need a free account to get an API key from them, but then yes, this script can now pull APRS reports and forward them on to all of the existing supported destinations.

closed 2024-06-18 by leoherzog

CT2IRJ, 2024-06-18:
Great work.
I'll try it when i get home.

Thanks again for your help

Salomão Fresco

Callsign: CT2IRJ

GRID Locator: IM59re -- CQ Zone 14 / ITU Zone 37
QTH: Fazendas de Almeirim, Portugal

CT2IRJ, 2024-06-20:
Hi again Leo,

I've tried briefly the new script, but it seems it is not working.
Maybe in the weekend (if kids allow) I'll give it another go.

Thanks

leoherzog, 2024-06-20:
Okay. If you could describe what steps you've done and where you got stuck, I'll do my best to help.