leoherzog/WundergroundStationForwarder #23: Literally need help

carcar367, 2024-06-18:
Hi so I tried to copy and paste the info in the code.gs file into google scripts and it just gave me a million error messages. If you want screenshots I'm happy to provide some. But like I need very detailed step-by-step intructions because I have little to no technical knowledge.

Thanks

leoherzog, 2024-06-18:
Hi @carcar367.

Here are some detailed instructions:

### 1. Create a Google Apps Script Project
1. Open your browser and go to [Google Apps Script](https://script.google.com/).
2. Click on **New Project**.
3. Name your project by clicking on the "Untitled project" name at the top and giving it any other new name.

### 2. Overwrite the Default Code
1. Delete any default code in the script editor.
2. Open [the `code.gs` file from this repository](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.6.0/code.gs).
3. Copy all the content of the `code.gs` file.
4. Paste it into the script editor in your new project.

### 3. Configure the Script
1. **Choose Your Datasource**:
   - Modify the `datasource` variable on line 10 according to your data source.
   - For example, for IBM Wunderground, set `datasource` to `ibm`.
   - Add the required API keys and station IDs as specified in the `code.gs` file.

2. **Configure Forwarding Destinations**:
   - Set the appropriate forwarding destinations to `true` and provide necessary API keys and IDs.
   - For example, to send data to Wunderground, set `updateWunderground` to `true` on line 32, and add your `wundergroundAPIKey` and `wundergroundStationId`.

### 4. Run the Schedule Function
1. In the script editor, select the `Schedule` function from the dropdown menu next to the `▷ Run` button.
2. Click on the `▷ Run` button to execute the function.
3. Grant any necessary permissions when prompted.

### 5. Monitor Executions
1. Click on the `Executions` tab on the left sidebar to monitor the script's periodic runs.

Does that help? What step(s) are you getting stuck on?

carcar367, 2024-06-18:
Not sure. I think I just fixed it...somehow. But coincidentially just as I came back to this I realized that there's a new release but I didn't install it because I'm not gonna go through all that pain just to put all my info in again. I think it's fixed now but I'll leave this open for a few more days to see if anything else comes up.

leoherzog, 2024-06-18:
Yeah, just added support for a new datasource and fixed a couple of bugs. If you want, you can copy/paste only everything below "Do Not Edit Below" ([Line 74](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.6.0/code.gs#L74) and down) to update, as long as you're not planning on using APRS (CWOP) as your datasource.

leoherzog, 2024-06-18:
I'm gonna close this since it's not an issue with the project, but feel free to continue commenting.

closed 2024-06-18 by leoherzog

carcar367, 2024-06-18:
Okay thank you. And does it only send every 15 minutes to CWOP?

leoherzog, 2024-06-18:
Every 5.

https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.6.0/code.gs#L113