leoherzog/WundergroundStationForwarder #24: ibm (wunderground) API keys

aolfish, 2024-07-02:
Weather Underground has changed they API key settings.  API keys are now limited to 1500 calls per day and 30 calls per minute.

That equals 1440 Calls per day for device.
I have 2 stations reporting so my daily call is 2880, which is over the limit .  
Weather Underground suspends my key and starts failing.
I regenerate new key and it will work for couple days then repeats.

Any way to change calling from 1 minute to 2 minutes ?

leoherzog, 2024-07-02:
Classic. Ugh.

Apps Script only lets you choose every one or five minutes :(

If you change [`Line 107`](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.6.0/code.gs#L107) in the code from:

```js
if (updateWunderground) ScriptApp.newTrigger('updateWunderground_').timeBased().everyMinutes(1).create();
```

to:

```js
if (updateWunderground) ScriptApp.newTrigger('updateWunderground_').timeBased().everyMinutes(5).create();
// options are every 1, 5, 10, 15 or 30 minutes
// https://developers.google.com/apps-script/reference/script/clock-trigger-builder#everyMinutes(Integer)
```

[image]

leoherzog, 2024-07-02:
There are alternative ways to schedule it every 2 minutes with a watcher function that sets exact every-two-minutes triggers, but it'd add a lot of complication to the code. Sorry 'bout that.

aolfish, 2024-07-02:
Thanks Leo. 
 May try directly from Acurite since you've updated since we first started.  That would cutdown half of my daily call.

karss79, 2024-07-04:
@aolfish: Instead of having two devices under one wunderground account, why don't you use separate WU account for each device. By this way, you may be able to overcome the WU API call limits.

leoherzog, 2024-08-08:
I think I'm gonna mark this issue as closed, since this is a limitation with the upstream provider and not with this project.

closed 2024-08-08 by leoherzog