leoherzog/WundergroundStationForwarder #25: TempestWX API response does not match code

josephgruber, 2024-09-07:
The response from the endpoint you're hitting, https://swd.weatherflow.com/swd/rest/observations/station/<STATION_ID>?token=<TOKEN>, does not match what you're trying to retrieve. For example, there is no `air_temperature` key in the `obs` entries. This causes failures when attempting to send to any provider as the `conditions` variable is practically blank with only the time and lat/long.
 
**Example response from Tempst**
```
{
elevation: 1234.567890,
indoor_keys: [
"timestamp",
"air_temperature_indoor",
"barometric_pressure_indoor",
"station_pressure_indoor",
"sea_level_pressure_indoor",
"pressure_trend_indoor",
"relative_humidity_indoor",
"precip_indoor",
"precip_accum_last_1hr_indoor",
"rain_accum_local_day_indoor",
"precip_accum_local_yesterday_final_indoor",
"precip_minutes_local_yesterday_final_indoor",
"wind_avg_indoor",
"wind_direction_indoor",
"wind_gust_indoor",
"wind_lull_indoor",
"solar_radiation_indoor",
"uv_indoor",
"brightness_indoor",
"lightning_strike_last_epoch_indoor",
"lightning_strike_last_distance_indoor",
"lightning_strike_count_indoor",
"lightning_strike_count_last_1hr_indoor",
"lightning_strike_count_last_3hr_indoor",
"feels_like_indoor",
"heat_index_indoor",
"wind_chill_indoor",
"dew_point_indoor",
"wet_bulb_temperature_indoor",
"wet_bulb_globe_temperature_indoor",
"delta_t_indoor",
"air_density_indoor"
],
is_public: true,
latitude: 39.1234,
longitude: -104.12345,
obs: [
{
air_density_indoor: 0.93165,
air_temperature_indoor: 24.9,
barometric_pressure_indoor: 797.1,
brightness_indoor: 22786,
delta_t_indoor: 12.2,
dew_point_indoor: 4.1,
feels_like_indoor: 24.9,
heat_index_indoor: 24.9,
lightning_strike_count_indoor: 0,
lightning_strike_count_last_1hr_indoor: 0,
lightning_strike_count_last_3hr_indoor: 0,
lightning_strike_last_distance_indoor: 41,
lightning_strike_last_epoch_indoor: 1725544504,
precip_accum_last_1hr_indoor: 0,
precip_accum_local_yesterday_final_indoor: 0,
precip_accum_local_yesterday_indoor: 0,
precip_analysis_type_yesterday_indoor: 0,
precip_indoor: 0,
precip_minutes_local_day_indoor: 0,
precip_minutes_local_yesterday_final_indoor: 0,
precip_minutes_local_yesterday_indoor: 0,
pressure_trend: "steady",
pressure_trend_indoor: "steady",
rain_accum_local_day_indoor: 0,
relative_humidity_indoor: 26,
sea_level_pressure_indoor: 1027.9,
solar_radiation_indoor: 190,
station_pressure_indoor: 797.1,
timestamp: 1725729252,
uv_indoor: 1.62,
wet_bulb_globe_temperature_indoor: 17,
wet_bulb_temperature_indoor: 12.7,
wind_avg_indoor: 1.4,
wind_chill_indoor: 24.9,
wind_direction_indoor: 274,
wind_gust_indoor: 2.2,
wind_lull_indoor: 0.6
}
],
outdoor_keys: [
"timestamp",
"pressure_trend",
"precip_minutes_local_day_indoor"
],
public_name: "Santa Claus Ln",
station_id: 123456,
station_name: "Home",
station_units: {
units_direction: "cardinal",
units_distance: "mi",
units_other: "imperial",
units_precip: "in",
units_pressure: "inhg",
units_temp: "f",
units_wind: "mph"
},
status: {
status_code: 0,
status_message: "SUCCESS"
},
timezone: "America/Denver"
}
```

josephgruber, 2024-09-07:
This was a misconfiguration on my device. It was set to indoor instead of outdoor.

closed 2024-09-07 by josephgruber

leoherzog, 2024-09-07:
:) glad you figured it out!