leoherzog/WundergroundStationForwarder #26: OpenWeatherMap station ID

esetleges, 2024-09-11:
Hi there,

Do you know how to get station ID for openweathermap? It seems its need some scripting in order to gain ID from their API. 

https://openweathermap.org/stations

Im not too familiar with coding, may any of you knows how to do this simply?

Thanks

esetleges, 2024-09-11:
Ok, I solved!!

closed 2024-09-11 by esetleges

leoherzog, 2024-09-11:
Nice work! I left an unused [`createNewOWMStation_()`](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L947-L974) function in there. If you remove the `_` from the name, you can run it from the `▷ Run` button in the menu bar. Perhaps I should make a "getOWMStations_()` function in there too, since the only way to interface with OpenWeatherMap stations is via their API :thinking:

esetleges, 2024-09-11:
You need POSTMAN request:

URL:
http://api.openweathermap.org/data/3.0/stations?&appid=xxxxxxc8fc462828fe9528e960f4da7e

(appid=API KEY from openweathermap)

HEADER:
Content-Type: application/json

BODY:
                    
{
  "external_id": "LOCATION_STATION_01",
  "name": "Sample Weather Station",
  "latitude": xx.xx,
  "longitude": xxx.xx,
  "altitude": number in meters
}

You get 200 response and the 'external_id' will be the station ID/name (for your script) next to api key.

I managed to get station id from phone with this: https://play.google.com/store/apps/details?id=com.app.restclient