leoherzog/WundergroundStationForwarder #27: Add MET (UK) support

esetleges, 2024-09-11:
It would be nice to have MET ((WOW) support too in the script:

https://wow.metoffice.gov.uk/

https://about.metservice.com/our-company/your-weather/add-your-own-weather-station

Cheers!

leoherzog, 2024-09-11:
Sounds cool! I'll read up on the docs.

leoherzog, 2024-09-11:
https://wow.metoffice.gov.uk/support/dataformats#automatic

Leaving here for reference.

closed 2024-09-16 by leoherzog

leoherzog, 2024-09-16:
Released [`v2.8.0`](https://github.com/leoherzog/WundergroundStationForwarder/releases/tag/v2.8.0) just now that adds this. Thanks again for the idea!

esetleges, 2024-09-25:
hey thanks.for the support. It seems the script runs to error with WOW:

Error	
Exception: Request failed for http://wow.metoffice.gov.uk returned code 400. Truncated server response: Bad Request (use muteHttpExceptions option to examine full response)
    at updateWOW_(Code:1053:30)

May they updating their servers ( notification on their site) or else.

commit 374b0e37 (2024-09-16): Add MET (UK) WOW Destination

Closes #27