leoherzog/WundergroundStationForwarder #29: Rain reported to Windy

eeeevl, 2024-09-16:
The plugin works a treat so far.  The one issue that I have come across is that the rainfall is not being reported to Windy from Wunderground.  I have all the other parameters working as they should its just not reporting the rainfall data.  Can someone help pls??
[image]

leoherzog, 2024-09-16:
Sure. It looks like [IBM lets you pull down current precipitation rate and total precipitation since midnight](https://docs.google.com/document/d/1KGb8bTVYRsNgljnNH67AMhckY8AQT2FVwZ9urj8SWBs/preview), whereas [Windy only accepts "precip from the last hour"](https://community.windy.com/topic/8168/report-your-weather-station-data-to-windy). I'm not sure there would be a way to calculate what Windy's expecting to receive given the stats that Wunderground gives out.

eeeevl, 2024-09-16:
Hey, Thanks for the reply...

Previously (whilst using CMX) Windy would display both hourly and also accumulated totals for a 24hr period though 🤷
[image]

jeffomoreira, 2024-09-16:
Hello.

Following this. I had not realized Windy really does not show rain data.

16 de set. de 2024, 10:55:01
Debug
{"update":{"update":{"stations":[],"observations":[{"si":0,"time":"2024-09-16T13:54:03.000Z","temp":12.833334360000002,"wind":1.43052432,"wind_dir":82,"wind_gust":2.86104864,"rh":85,"dew_point":50.6,"pressure":102844.564079}]},"errors":{"observations":[],"stations":[]}},"result":{"0":{"stations":[],"observations":[{"success":true,"value":{"si":0,"time":"2024-09-16T13:54:03.000Z","temp":12.833334360000002,"wind":1.43052432,"wind_dir":82,"wind_gust":2.86104864,"rh":85,"dew_point":50.6,"pressure":102844.564079}}]}}}

Refresh from WU
16 de set. de 2024, 11:04:49
Depuração
{"time":1726495480000,"latitude":"-25.516092","longitude":"-49.195479","temp":{"f":55.3,"c":12.94},"dewpoint":{"f":50.8,"c":10.44},"windSpeed":{"mph":1.6,"mps":0.72,"kph":2.57,"knots":1.39},"windGust":{"mph":3.3,"mps":1.48,"kph":5.31,"knots":2.87},"winddir":212,"windChill":{"f":55.3,"c":12.94},"heatIndex":{"f":55.3,"c":12.94},"pressure":{"inHg":30.37,"hPa":1028},"humidity":85,"precipRate":{"in":0,"mm":0},"precipSinceMidnight":{"in":1.05,"mm":26.67}}

Thanks.

leoherzog, 2024-09-16:
Because [Windy only supports taking in "precipitation over the last hour" measurements](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L831), data sources that don't report that data point such as [IBM Wunderground](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L182-L189), [MyAcurite](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L340-L343), [Ambient Weather](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L588-L599), or [APRS](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L661-L672) do not support sending precipitation data to Windy. [Davis Weatherlink](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L441-L444) and [Weatherflow Tempest](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.7.0/code.gs#L524-L527) report that correctly and work as expected.

I'll have to put some thought into this. Each of the currently-unsupported services report a _current_ precipitation rate, so if we can keep a rolling 1-hour average in the Apps Script cache that's continually updated, then in theory we could start to send this on to Windy.

jeffomoreira, 2024-09-16:
Hi Leo.

Good thoughts, that might work.
I only noticed today after eeevl's email, that Windy, is, indeed, not
showing the rain data. And we hat a LOT of rain here last day.

leoherzog, 2024-09-16:
Alrighty, @eeeevl and @jeffomoreira. I've just published `v2.7.1` which I _think_ does this correctly. There's a new calculation in the code that makes some assumptions based off of the last hour minutely history of current precipitation rates to try and come up with a best estimate of the

jeffomoreira, 2024-09-16:
Thanks, Leo. And eevil.

I updated my script a couple hours ago but for some reason rain is still
not showing up. Will look into it calmly when I get some time.

jeffomoreira, 2024-09-19:
Rain to Windy seems to be working! Looks fine to me. Thanks guys.

[image: image.png]

eeeevl, 2024-09-20:
Rain being reported to Windy as well - slightly inaccurate (0.3mm) but appears to be working fine...