leoherzog/WundergroundStationForwarder #31: Station Forwarder ... newbie issues

leorecski, 2024-10-03:
Hi there. I followed exactly the steps you mentioned to send my data to Windy, Windguru and WU. Unfortunately I am getting error messages concerning 3 lines. Could you help me solve the issue and get the data up running. Your hjelp is much appreciated. Regards, Leonardo.

[image]

eeeevl, 2024-10-03:
I thought that I might be the only one with a problem, however this just started happening to me as well recently...  Different lines however (I assume because I am pulling it from WU and not WOW)

[image]

chiefcomm, 2024-10-03:
> Hi there. I followed exactly the steps you mentioned to send my data to Windy, Windguru and WU. Unfortunately I am getting error messages concerning 3 lines. Could you help me solve the issue and get the data up running. Your hjelp is much appreciated. Regards, Leonardo.
> 
> ![Screenshot_1](https://private-user-images.githubusercontent.com/72770874/373256808-850ced4e-c2dc-40cf-9309-14af7500433f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5OTk4ODgsIm5iZiI6MTcyNzk5OTU4OCwicGF0aCI6Ii83Mjc3MDg3NC8zNzMyNTY4MDgtODUwY2VkNGUtYzJkYy00MGNmLTkzMDktMTRhZjc1MDA0MzNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAzVDIzNTMwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg5NTgxMGE5N2M2N2ExNzk2ZjQwNjNmYzBiZjViNTJhYTBkZTI1M2M4ZTQzZTIyOGVhMjZiNzU3ZDRkODY0NmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pFbk6SKu3XTyUMF0SvTIlEh31_ukjJcqnO5y3S8YMKM)

no issues on my side - I pull from Tempest Weatherflow and push to Wunderground, PWSWeather & OpenWeatherMap no errors reported by any of the uploads.   Maybe check your station ID and API token.

eeeevl, 2024-10-04:
My station ID and APIs haven't changed in weeks.  It has only just become a problem now for some reason.   Seems like it times out or something but also get these messages from both upload and download locations...

[image]

chiefcomm, 2024-10-04:
> My station ID and APIs haven't changed in weeks. It has only just become a problem now for some reason. Seems like it times out or something but also get these messages from both upload and download locations...
> 
> <img alt="image" width="1187" src="https://private-user-images.githubusercontent.com/181693572/373448391-33c3e695-be1f-4b98-8528-39eddbf9584d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgwMDA3ODksIm5iZiI6MTcyODAwMDQ4OSwicGF0aCI6Ii8xODE2OTM1NzIvMzczNDQ4MzkxLTMzYzNlNjk1LWJlMWYtNGI5OC04NTI4LTM5ZWRkYmY5NTg0ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAwNFQwMDA4MDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lODZlMzU3MGMwNTc1NjZhNmY2OTE4MGExZDk3Y2IzNTVmMDI3MTY2NjNhYTk2ZjdhYzIyZWI3MWE0MmEwNGRiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.JUHvlzircC_l_9uhgUqNbett2e5uEWCRxiYoG6DTP-E">

Sorry I was responding to @leorecski as I have similar setup.  I cannot comment or offer insight pulling from WU.

eeeevl, 2024-10-04:
Sorry, my bad :)

leoherzog, 2024-10-06:
@leorecski Sorry to hear you're having trouble. If I'm understanding correctly, it hasn't ever worked properly. In that case, consider triple-checking that the Station ID and [Personal Access Token](https://tempestwx.com/settings/tokens) are correct on lines 25 and 24. It should be enclosed in quotes with no spaces within the quotes, e.g.

```js
const weatherflowPUT = '1234abcd-4321-dcba-1234-abcd1234dcba';
const weatherflowStationId = '18150';
```

eeeevl, 2024-10-08:
@leoherzog what would be causing my issue then.  FYI it also just randomly started working again a few days ago.  I'm at a loss to explain this (as I know my station details and APIs are 100% correct).  Very curious indeed

leoherzog, 2024-10-08:
@eeeevl Is your Wunderground API Key expired? 

https://www.wunderground.com/member/api-keys

eeeevl, 2024-10-08:
No the keys are all current. Like I said though, it's back working as per the norm and I didn't change anything. It's got me scratching my head...

leorecski, 2024-10-08:
Perhaps my problem is that I do not have a Weatherflow account, so I cannot supply the required info for lines 24 and 25. I just realized that. So, I guess I cannot use the code.

leoherzog, 2024-10-08:
@eeeevl Sounds like a temporary outage with IBM and their Wunderground API.

leoherzog, 2024-10-08:
@leorecski You'll need to pull readings from _somewhere_ if you want to forward them to Windy, Windguru and WU. What kind of station were you wanting to pull readings from?

leorecski, 2024-10-08:
> @leorecski You'll need to pull readings from _somewhere_ if you want to forward them to Windy, Windguru and WU. What kind of station were you wanting to pull readings from?

I am pulling from WU. Do I need to change line 10 and put 'wunderground' instead of 'weatherflow'? Am I that stupid?

leoherzog, 2024-10-08:
@leorecski It sounds like you want to take data _from_ Wunderground (setting [line 10](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.0/code.gs#L10) to `ibm` and [lines 12-13](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.0/code.gs#L12-L13) with your Wunderground info) and to send data _to_ Windy and WindGuru ([lines 42-44](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.0/code.gs#L42-L44) and [lines 59-61](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.0/code.gs#L59-L61))?

leorecski, 2024-10-08:
Yes that was the idea. Use the data my station sends to WU and from there, using the code send to Windy and Windguru. So lines 12-13 were filled with my credentials from WU. I did not know I should change amything on line 10. On the sending data lines I added info for Windy, Windguru and again to WU.

leoherzog, 2024-10-08:
You don't want to enable sending to WU. Only receiving from it.

leorecski, 2024-10-08:
Got it. Will try again.

leorecski, 2024-10-08:
Well, same problem. On lines 42 and 59 do I have to change 'false' to 'true' for Windy and Windguru?
[image]

leoherzog, 2024-10-08:
Yes. You still have [line 10](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.1/code.gs#L10) set to `weatherflow`, not `ibm`. You want to be pulling readings from IBM Wunderground.

And yes, you need to set [line 42](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.1/code.gs#L42) and [line 59](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.1/code.gs#L59) to `true` to start updating those services.

Please read [the README documentation in this repository](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.1/README.md) carefully for these setup instructions.

jeffomoreira, 2024-10-12:
Hey Leo.

I know there may be a few discrepancies on sending and converting rain data
from WU to Windy.

Today we had a considerable amount of rain here. Wu reports (24h):
[image: image.png]
While Windy is showing:
[image: image.png]
Roughly half of the numbers from WU.

I am just reporting in hope of, maybe, helping to calibrate/adjust the data
reported/calculated and sent to Windy.

Thanks.
Jeffo.

rubengte, 2024-10-12:
I believe this script Leo created is just brilliant.
Truly deeply thankful to have found you and this gift 🙏

My suggestion is the same I sent to Wunderground, (without receiving any feedback to date), which is to find some kind of filter to exclude aberration readings like for example when the station shows a peak out of nowhere from 100 or sometimes even 400km/h 

The same happens with rain also. 
Fortunately it's not often, however it happens. 

If there's something that can be included to filter these events out that would also be amazing ✨

Best,
Rúben

leoherzog, 2024-10-14:
@jeffomoreira Hi Jeffo. It looks like your attachments didn't arrive via email → Github properly. Perhaps come to the Github web interface and try adding them there?

https://github.com/leoherzog/WundergroundStationForwarder/issues/31#issuecomment-2408315658

leoherzog, 2024-10-14:
@rubengte Sure, it should be possible. What datasource are you using, and what are the upper limits that you'd like to ignore? If the number that the datasource returns is higher than your predefined ceiling, what do you want to happen? Clamp it to the ceiling, or set it to 0, or something else?

rubengte, 2024-10-14:
> @rubengte Sure, it should be possible. What datasource are you using, and what are the upper limits that you'd like to ignore? If the number that the datasource returns is higher than your predefined ceiling, what do you want to happen? Clamp it to the ceiling, or set it to 0, or something else?

That would be amazing. 
I'm getting the data from Wunderground to windy and to Windguru.
Not sure about setting a specific amount, i would prefer (if possible) to set a compare to previous, i attached an image to better illustrate what i mean.

If this is not possible we could use 100km/h for the wind and maybe 50mm for the rain however, this site can (although rarely) reach 100km/h wind.

What we could do is ignore the reading and wait for the next or repeat the previous until it's back to "normal", something like this, what do you think?

These "aberration" readings happen every week, at least once, and it messes all the graph readings, so it would be really super to filter them out.

P.s. i would like to contribute also to the project, you're doing a fantastic work and I would like to send a contribution, however can't find a way.
Do you have revolut? 

Thanks again Leo

[image]

leoherzog, 2024-10-15:
Moved to new issue (#33) and closing this issue, as we've definitely changed the scope of the original request in this issue.

closed 2024-10-15 by leoherzog