leoherzog/WundergroundStationForwarder #32: Difference between rain: Windy from WU

jeffomoreira, 2024-10-14:
Hey Leo. (re-sending the message)

I know there may be a few discrepancies on sending and converting rain data from WU to Windy.

Today we had a considerable amount of rain here. Wu reports (24h):
[image]

While Windy is showing:
[image]

Roughly half of the numbers from WU.

I am just reporting in hope of, maybe, helping to calibrate/adjust the data reported/calculated and sent to Windy.

Thanks.
Jeffo.

leoherzog, 2024-10-15:
Interesting.

Well, Wunderground may be receiving rainfall rate data as frequently as every 2.5 seconds with [their RapidFire system](https://support.weather.com/s/article/PWS-Upload-Protocol?language=en_US). As [I mentioned before](https://github.com/leoherzog/WundergroundStationForwarder/issues/29#issuecomment-2353155561), Apps Script can only ask Wunderground for new data as frequently as every minute (without getting into complicated multiple-trigger systems with locking, etc etc). [The hack](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.1/code.gs#L1108-L1136) for calculating rain accumulations over the last hour for services that don't provide them (like Wunderground) is to take the history of current rates (as received every minute) over the last hour and total them.

Once you have a full hour of minutely rain rates as they were during those moments, you can add them up and get an approximate total over the hour. So, if your weather station is reporting to Wunderground every 2.5 seconds, and you get a sudden 30 second downburst that starts and stops, Wunderground will record that, but if Apps Script doesn't happen to pull for another 30 seconds and ask for the current rain rate at the right time, it will miss that.

I _guess_ that's what happened here? If this is true, then the opposite could also happen; We could ask Wunderground for the current rate right during the middle of a really short burst, and we might assume that that lasted longer than it did.

Keep an eye on the discrepancy over the next few rainstorms and report back if you do or do not see the rate on Windy be reported as _more_ than Wunderground. That will help confirm these suspicions. If this is indeed what's happening, over longer periods and multiple rain events, these discrepancies should theoretically average out.

jeffomoreira, 2024-10-15:
Hey.

I got your point and it makes sense. I have a hard time with my abstraction logic capability these days, but I understand that the number in WU may be being "diluted" over time, on Windy.

Another thing I might have noticed is that the accumulated rain may have being reset in the middle of the day, on Windy. In that case Windy should only reset at 00h/Midnight...isn't it?. May be another think to check out? The pictures below bring some sense to it: near 10AM you see "24h acc: 8.1mm"and then after 1PM (13:38) "24h acc: 7.3mm". The accumulated values, over the day, should never reduce, only increase, right?

See the pictures:
[image]

[image]

Hope that helps.
Thanks.

jeffomoreira, 2024-10-16:
> As I look at your graphs, I see what I expect to see. I think that's how
> all other stations also behave?
>
> Here's mine:
> https://www.windy.com/station/pws-f069ecca

Yes, they are looking good, rain is being displayed but it wasn't clear for
me that the Wunderground 24h accumulated rain was, in the example, 15.4mm
and the 24 acc in Windy was only 7. That was my concern.

Friday, 11th,  15.4mm WU:
https://www.wunderground.com/dashboard/pws/ISOJOS39/graph/2024-10-11/2024-10-11/daily

Friday, 11th,7mm, Windy:
https://www.windy.com/station/pws-f052cfed?-26.623,-49.197,8\

To me it is no big deal since what matters most is what is going from WU to
MADIS/NOAA (which is correct), but I was curious about why Windy shows it
differently, half of what WU sends.

leoherzog, 2025-04-02:
@jeffomoreira How's your data been over the last few months? Think we can close this issue?