leoherzog/WundergroundStationForwarder #34: Help me I know nothing

Hero21b, 2024-10-29:
Here is my error message:

Error	
Problem fetching https://swd.weatherflow.com/swd/rest/observations/station/xxxxx?token=xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx
fetchJSON_	@ Code.gs:1192
refreshFromWeatherflow_	@ Code.gs:477
Schedule	@ Code.gs:102

The only edits I have made are to my Wunderground API + Station ID, and to  updateWindy = true and the Windy API Key. I am useless with coding

[image]
[image]

Hero21b, 2024-10-29:
Nevermind, I see that I needed to update weatherflow to ibm on line 10. I guess I'm not fully useless, just slow lol.

closed 2024-10-29 by Hero21b

leoherzog, 2024-10-29:
Are you trying to pull station readings from Wunderground and send them only to windy.com? If so, you'll also need to change line 10 to

```js
const datasource = 'ibm';
```

As you have it now, it's still trying to pull readings from Weatherflow, which is a different weather station brand.

Hero21b, 2024-10-29:
Thanks. The reason I want to use this is that I have an Ecowitt station which can't send directly to Windy. Thanks so much for building this!

leoherzog, 2024-10-29:
Sure thing! Looks like we commented at the same time 😅 have a good open