leoherzog/WundergroundStationForwarder #37: Windy Upload Error 400

CaveJohnson10, 2024-12-12:
I get this error when trying to upload data to Windy.

Dec 11, 2024, 8:20:04 PM	Error	Exception: Request failed for https://stations.windy.com returned code 400. Truncated server response: {"observations":[],"stations":[]} (use muteHttpExceptions option to examine full response)
    at updateWindy_(Code:865:30)

I double checked my inputs to make sure everything is entered correctly. All the other functions complete successfully.

Any idea what is causing the issue?

leoherzog, 2024-12-12:
Hmmm. A couple of thoughts.

1. Make sure that you're properly specifying your `windyAPIKey` as the whole key, within quotes, and the `windyStationId` as `"0"` (or if you have multiple, the correct ID number).

https://stations.windy.com/stations
↓
https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.2/code.gs#L42-L44

2. If you're sure those are correct, let's add some additional logging to the `updateWindy_()` function.

https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.8.2/code.gs#L847-L871
↓
```js
// https://community.windy.com/topic/8168/report-your-weather-station-data-to-windy
function updateWindy_() {
  
  let conditions = JSON.parse(CacheService.getScriptCache().get('conditions'));

  console.log(JSON.stringify(conditions)); // add this
  
  let request = 'https://stations.windy.com/pws/update/' + windyAPIKey;
  request += '?stationId=' + windyStationID;
  request += '&time=' + new Date(conditions.time).toISOString();
  if (conditions.temp != null) request += '&tempf=' + conditions.temp.f;
  if (conditions.dewpoint != null) request += '&dewptf=' + conditions.dewpoint.f;
  if (conditions.windSpeed != null) request += '&windspeedmph=' + conditions.windSpeed.mph;
  if (conditions.windGust != null) request += '&windgustmph=' + conditions.windGust.mph;
  if (conditions.winddir != null) request += '&winddir=' + conditions.winddir;
  if (conditions.pressure != null) request += '&baromin=' + conditions.pressure.inHg;
  if (conditions.humidity != null) request += '&humidity=' + conditions.humidity;
  if (conditions.precipLastHour != null) request += '&rainin=' + conditions.precipLastHour.in;
  if (conditions.uv != null) request += '&uv=' + conditions.uv;

  console.log(request); // and this
  
  let response = UrlFetchApp.fetch(request).getContentText();
  
  console.log(response);
  
  return response;
  
}
```

CaveJohnson10, 2024-12-13:
Unfortunately, this did not solve the issue and I am still receiving the exact same error.

leoherzog, 2024-12-13:
Sorry, I should have been more clear. Now that those have been added, what do you see in the `☰▶` Execution tab logs every time `updateWindy_()` runs?

leoherzog, 2025-01-14:
@CaveJohnson10 Still having the issue?

CaveJohnson10, 2025-01-20:
Hi, Yes the exact same error occurs.

leoherzog, 2025-01-20:
Is Windy the only one that's failing? Do you have other upload destinations that are working?

CaveJohnson10, 2025-01-20:
I just tried Weather Underground and the upload itself to WU is successful in the script BUT it doesn't send any actual data. On the station page on WU I can see that that it says Online (Updated 1 minute ago) and it updates every minute but there's no weather data showing. I am starting to think there is an issue with the information the script if fetching from Davis Weatherlink.

leoherzog, 2025-01-20:
That's my suspicion as well.

CaveJohnson10, 2025-01-21:
Were you going to look further into the issue?

joseiaran, 2025-02-25:
7:50:48	Información	{"time":1740480600000,"latitude":"-38.20989","longitude":"-59.87107","dewpoint":{"f":68.99,"c":20.55},"windSpeed":{"mph":10,"mps":4.47,"kph":16.09,"knots":8.69},"windGust":{"mph":15,"mps":6.71,"kph":24.14,"knots":13.03},"winddir":323,"pressure":{"inHg":29.773,"hPa":1008},"humidity":94,"windChill":{"f":68.71,"c":20.4},"heatIndex":{"f":73.3,"c":22.94},"solarRadiation":104,"precipRate":{"in":0,"mm":0},"precipSinceMidnight":{"in":null,"mm":null},"precipLast24Hours":{"in":null,"mm":null},"precipLastHour":{"in":null,"mm":null}}

leoherzog, 2025-02-26:
@joseiaran Interesting. Can you uncomment Line 427 (by removing the `//`), save, and tell me what you see in the logs?

https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.9.5/code.gs#L427

joseiaran, 2025-03-01:
Hi:

11:54:54	Información	{"station_id":140262,"station_id_uuid":"1a67f356-ffda-4d5c-99c6-d584aca23f94","sensors":[{"lsid":541529,"sensor_type":82,"data_structure_type":2,"data":[{"ts":1740840600,"tz_offset":-10800,"bar_trend":60,"bar":29.764,"temp_in":81.4,"hum_in":56,"temp_out":76.2,"wind_speed":9,"wind_speed_10_min_avg":9,"wind_dir":203,"temp_extra_1":null,"temp_extra_2":null,"temp_extra_3":null,"temp_extra_4":null,"temp_extra_5":null,"temp_extra_6":null,"temp_extra_7":null,"temp_soil_1":null,"temp_soil_2":null,"temp_soil_3":null,"temp_soil_4":null,"temp_leaf_1":null,"temp_leaf_2":null,"temp_leaf_3":null,"temp_leaf_4":null,"hum_out":79,"hum_extra_1":null,"hum_extra_2":null,"hum_extra_3":null,"hum_extra_4":null,"hum_extra_5":null,"hum_extra_6":null,"hum_extra_7":null,"rain_rate_clicks":0,"rain_rate_in":0,"rain_rate_mm":0,"uv":null,"solar_rad":736,"rain_storm_clicks":206,"rain_storm_in":1.6220472,"rain_storm_mm":41.2,"rain_storm_start_date":null,"rain_day_clicks":80,"rain_day_in":0.62992126,"rain_day_mm":16,"rain_month_clicks":80,"rain_month_in":0.62992126,"rain_month_mm":16,"rain_year_clicks":901,"rain_year_in":7.094488,"rain_year_mm":180.2,"et_day":0.033,"et_month":0,"et_year":1.235,"moist_soil_1":null,"moist_soil_2":null,"moist_soil_3":null,"moist_soil_4":null,"wet_leaf_1":null,"wet_leaf_2":null,"wet_leaf_3":null,"wet_leaf_4":null,"forecast_rule":120,"forecast_desc":"Clearing, cooler and windy.","dew_point":69.20776531392512,"heat_index":78.49600000000001,"wind_chill":74.74809921120578,"wind_gust_10_min":14}]}],"generated_at":1740840894}
11:54:54	Información	{"time":1740840600000,"latitude":"-38.20989","longitude":"-59.87107","dewpoint":{"f":69.21,"c":20.67},"windSpeed":{"mph":9,"mps":4.02,"kph":14.48,"knots":7.82},"windGust":{"mph":14,"mps":6.26,"kph":22.53,"knots":12.17},"winddir":203,"pressure":{"inHg":29.764,"hPa":1008},"humidity":79,"windChill":{"f":74.75,"c":23.75},"heatIndex":{"f":78.5,"c":25.83},"solarRadiation":736,"precipRate":{"in":1.622,"mm":41.2},"precipSinceMidnight":{"in":null,"mm":null},"precipLast24Hours":{"in":null,"mm":null},"precipLastHour":{"in":null,"mm":null}}

joseiaran, 2025-03-01:
The windy station does not show temperature. Wind, Hum and Pressure show but no show temperature. Change line 427, same problem not show temperature. 

Thank's for you help.

leoherzog, 2025-03-03:
@CaveJohnson10 @joseiaran Thanks for the logs, it helped me find the issue (I think). Give [v2.9.6](https://github.com/leoherzog/WundergroundStationForwarder/releases/tag/v2.9.6) a try and let me know if it fixes it!

CaveJohnson10, 2025-03-03:
Thanks for looking into this. I have upgraded my Davis Weather Station to Pro+ for another year as I wasn’t sure if this issue would be resolved quickly. I’ll definitely try this uploader again next year once my subscription expires.

leoherzog, 2025-04-02:
@CaveJohnson10 I have rewritten the Davis receiver to attempt to be more flexible with possible values that the Davis API might return (issue #46). Think you could give this testing version a try and see if it fixes your issue?

https://github.com/leoherzog/WundergroundStationForwarder/blob/davis-fix/code.gs

leoherzog, 2025-07-07:
I believe that this is now fixed with [`v2.10.0`](https://github.com/leoherzog/WundergroundStationForwarder/releases/tag/v2.10.0). Give 'er a try and let me know how it goes!

closed 2025-07-07 by leoherzog