leoherzog/WundergroundStationForwarder #39: Current temperature failure in Windy

ferherlo, 2024-12-15:
Hi, Windy displays the data for Ecowitt station, but it shows the day's maximum temperature, not the current temperature. What could be the error?

Thank you

ferherlo, 2024-12-15:
The weather station is https://www.windy.com/station/pws-f04c24ce?40.885,-1.245,12

ferherlo, 2024-12-15:
Information

{"time":1734289634000,"temp":{"f":47.7,"c":**8.72**},"dewpoint":{"f":39.8,"c":4.33},"humidity":74,"windSpeed":{"mph":2,"mps":0.89,"kph":3.22,"knots":1.74},"windGust":{"mph":2.2,"mps":0.98,"kph":3.54,"knots":1.91},"winddir":102,"pressure":{"inHg":30.47,"hPa":1032},"solarRadiation":0,"uv":0,"precipRate":{"in":0,"mm":0},"precipSinceMidnight":{"in":0,"mm":0},"precipLastHour":{"in":0,"mm":0},"windChill":{"f":47.7,"c":8.72},"heatIndex":{"f":101.74,"c":38.75}}

20:07:18
Información
Scheduled! Check Executions ☰▶ tab for status.

ferherlo, 2024-12-15:
I think it’s already fixed. It was pulling data from another station I have at a different location

leoherzog, 2024-12-16:
Whoops! That'll do it :)

closed 2024-12-16 by leoherzog