leoherzog/WundergroundStationForwarder #40: Upload to Wunderground failed after script update to v2.9.0

chiefcomm, 2024-12-15:
Hi All, hopefully someone can point out my error.  I've been using the script for so long I can't remember and today I decided to update the script to the latest v2.9.0.

Since updating to v2.9.0 I have the following error in logs uploading to Wunderground (uploading to PWSWeather is OK).

[image]

Any clues? - I checked the Station ID and Key which have remained unchanged in years.

chiefcomm, 2024-12-16:
Ignore this I linked WeatherFlow to Wunderground via the native WeatherFlow integration.

closed 2024-12-16 by chiefcomm

leoherzog, 2024-12-16:
Hmm. If you were updating from 1.x.x, some of the variable names at the top of the document might have changed. Make sure that the Lines [10-73](https://github.com/leoherzog/WundergroundStationForwarder/blob/v2.9.0/code.gs#L10-L73) don't have any missing or misnamed variables!