leoherzog/WundergroundStationForwarder #41: Rain Accumulation is being reported as "Rain Rate" from 'acurite'

skysummit, 2024-12-27:
When pulling data from myAcurite, rain accumulation is not being pulled and is forwarding rainfall as "rain rate" to multiple websites.

EX:
[image]

{"time":1735316423000,"latitude":"30.418869","longitude":"-90.421232","temp":{"f":74,"c":23.33},"dewpoint":{"f":71,"c":21.67},"windSpeed":{"mph":5,"mps":2.24,"kph":8.05,"knots":4.34},"windGust":{"mph":9,"mps":2.5,"kph":14.48,"knots":7.82},"winddir":232,"pressure":{"inHg":30.03,"hPa":1017},"humidity":90,"windChill":{"f":74,"c":23.33},"heatIndex":{"f":72.3,"c":22.38},"precipRate":{"in":0.01,"mm":0.25},"precipLastHour":{"in":0.01,"mm":0.25}}

leoherzog, 2024-12-31:
I'm having a hard time recreating this.

```json
{
  "1": [
    {
      "happened_at": "2024-12-31T00:02:23+00:00",
      "raw_values": {"C": 3.3, "F": 37.9}
    },
    {
      "happened_at": "2024-12-31T01:02:23+00:00",
      "raw_values": {"C": 2.4, "F": 36.3}
    },
    {
      "happened_at": "2024-12-31T02:02:22+00:00",
      "raw_values": {"C": 2.5, "F": 36.5}
    },
    {
      "happened_at": "2024-12-31T03:02:24+00:00",
      "raw_values": {"C": 0.7, "F": 33.3}
    },
    {
      "happened_at": "2024-12-31T04:02:22+00:00",
      "raw_values": {"C": 0.2, "F": 32.4}
    },
    {
      "happened_at": "2024-12-31T05:02:22+00:00",
      "raw_values": {"C": 0.5, "F": 32.9}
    },
    {
      "happened_at": "2024-12-31T06:02:23+00:00",
      "raw_values": {"C": -0.1, "F": 31.8}
    },
    {
      "happened_at": "2024-12-31T07:02:23+00:00",
      "raw_values": {"C": 0.1, "F": 32.2}
    },
    {
      "happened_at": "2024-12-31T08:02:22+00:00",
      "raw_values": {"C": 0.6, "F": 33.1}
    },
    {
      "happened_at": "2024-12-31T09:02:22+00:00",
      "raw_values": {"C": 1.0, "F": 33.8}
    },
    {
      "happened_at": "2024-12-31T10:02:22+00:00",
      "raw_values": {"C": 1.4, "F": 34.5}
    },
    {
      "happened_at": "2024-12-31T11:07:04+00:00",
      "raw_values": {"C": 1.7, "F": 35.1}
    },
    {
      "happened_at": "2024-12-31T12:02:22+00:00",
      "raw_values": {"C": 1.9, "F": 35.4}
    },
    {
      "happened_at": "2024-12-31T13:02:23+00:00",
      "raw_values": {"C": 2.0, "F": 35.6}
    },
    {
      "happened_at": "2024-12-31T14:02:22+00:00",
      "raw_values": {"C": 1.8, "F": 35.2}
    },
    {
      "happened_at": "2024-12-31T15:02:22+00:00",
      "raw_values": {"C": 1.8, "F": 35.2}
    },
    {
[... 864 lines cut ...]
      "last_reading_value": "5",
      "channel": 12,
      "chart_unit": "mph",
      "wind_speed_peak_value": null,
      "display_unit": "mph"
    }
  ],
  "wired_sensors": [],
  "model": {
    "id": 4020,
    "code": "5in1WS",
    "description": "AcuRite Iris®",
    "has_wifi": false
  }
}
```

My device reports three separate rainfall values in its hourly summaries:

1. Channel 10 appears to be hourly/instantaneous readings
2. Channel 11 shows accumulated values that reset at 5:02 AM

The real-time last_reading_value matches Channel 10's readings... In the example data, both channels show 0.01" at 18:02, matching the real-time endpoint's `last_reading_value: "0.01"`. So I think that in my case right now, it's reporting an instantaneous rain rate as currently written? It's so hard to know for sure since this is simulating being the MyAcurite website, using an undocumented API.

Is it possible that you've somehow misread your side and that it's actually currently correct?

leoherzog, 2024-12-31:
Hold on. It's currently raining here in West Michigan... and I may be able to track this over the next hour to confirm!

leoherzog, 2024-12-31:
Okay. Here's a potential fix. I think I re-created the issue.

https://github.com/leoherzog/WundergroundStationForwarder/commit/eb6ba17d7f86e7e90bd0edc6f8fba1726b1da64b

This is the change I made. I'll try running this beta version of this function for a little bit:

https://github.com/leoherzog/WundergroundStationForwarder/blob/eb6ba17d7f86e7e90bd0edc6f8fba1726b1da64b/code.gs#L221-L399

If you feel so led, try 'er out @skysummit.

skysummit, 2024-12-31:
> eb6ba17

Ok thanks!

It actually may be reporting correctly today because I have myAcurite
reporting to WU and then using the WU API to pull from. Ever since I
switched to that, I believe it's been correct.

Is there any way to possibly calibrate a reading within the script before
it reports to CWOP, etc?