leoherzog/WundergroundStationForwarder #44: Wx not appear in Windy

Sv1jmo, 2025-01-14:
Hi 
suddenly stopped presenting the station before 2 days . i update the code , i check everthing .
is any general issue ? have others the same problem? 

Execution log   ( looks ok ) 
10:04:48 AM	Notice	Execution started
10:04:49 AM	Info	{"time":1736841882000,"latitude":"38.928825","longitude":"20.85182","temp":{"f":44.2,"c":6.78},"dewpoint":{"f":34.1,"c":1.17},"windSpeed":{"mph":8.9,"mps":3.98,"kph":14.32,"knots":7.73},"windGust":{"mph":10.5,"mps":4.69,"kph":16.9,"knots":9.12},"winddir":135,"windChill":{"f":39.3,"c":4.06},"heatIndex":{"f":44.2,"c":6.78},"pressure":{"inHg":30.18,"hPa":1022},"humidity":68,"precipRate":{"in":0,"mm":0},"precipSinceMidnight":{"in":0.22,"mm":5.59}}
10:04:51 AM	Info	Scheduled! Check Executions ☰▶ tab for status.
10:04:51 AM	Notice	Execution completed

ferherlo, 2025-01-14:
I have the same problem. Everything in the code appears to be correct, and the submission is also correct. It just stopped working

Sv1jmo, 2025-01-14:
no errors

ferherlo, 2025-01-14:
> Hmm. My Windy submissions are still showing. Are there any errors in the log under the `updateWindy_()` executions?

No, it´s correct. Any errors:

14 ene 2025, 22:13:32
Depuración
{"update":{"update":{"stations":[],"observations":[{"si":0,"time":"2025-01-14T21:08:42.000Z","temp":-6.000000480000001,"wind":0.581150505,"wind_dir":110,"wind_gust":0.9834854700000001,"rh":84,"dew_point":-8.27777844,"pressure":102912.291813,"precip":0,"uv":0}]},"errors":{"observations":[],"stations":[]}},"result":{"0":{"stations":[],"observations":[{"success":true,"value":{"si":0,"time":"2025-01-14T21:08:42.000Z","temp":-6.000000480000001,"wind":0.581150505,"wind_dir":110,"wind_gust":0.9834854700000001,"rh":84,"dew_point":-8.27777844,"pressure":102912.291813,"precip":0,"uv":0}}]}}}

ferherlo, 2025-01-14:
Upsss

In this link the data does appear: https://www.windy.com/station/pws-f04c24ce?40.908,-1.299,16
But not in the preview with the temperature map: https://www.windy.com/?40.908,-1.299,16

leoherzog, 2025-01-14:
So what I think I'm hearing you say is, "I used to see my weather station data on [the Windy.com map's temperature layer](https://www.windy.com/-Temperature-temp), but now I can only see it when I browse directly to my weather station's link". Is that right?

You can see your weather station's URL by clicking the 'See at windy.com `□↗`' button next to your weather station on [the stations management page](https://stations.windy.com/stations)].

If that's the case, then I have no control over how windy.com decides to show weather station data on their maps.

ferherlo, 2025-01-14:
Yes, that's the case. I don't understand why they hide the data on the Windys maps temperature layer. It's frustrating.

leoherzog, 2025-01-14:
Perhaps consider discussing it and/or submitting feedback on the [Windy Community Forum](https://community.windy.com/).

closed 2025-01-14 by leoherzog