leoherzog/WundergroundStationForwarder #46: Windy no rain with Davis Weatherlink

joseiaran, 2025-03-30:
Hi, the new problem found, in the windy data station no view rain in day. i'm see Wind, Temperature, Presurre and Humidity. No view Rain. 

Thanks For you help, the script is excelentt.
Ignacio

leoherzog, 2025-03-31:
Hello @joseiaran !

Someone else on a forum told me that they were also having issues with Davis Weatherlink. Apparently their API can return different things depending on the _model_ of the Weather Station (e.g. the same API returning `temp` vs. `temp_out`).

To attempt to fix this, I've made a tweak that I need help testing. Can you try this version and let me know if it works? 

https://github.com/leoherzog/WundergroundStationForwarder/blob/davis-fix/code.gs

joseiaran, 2025-03-31:
Hi' i'm try the night the new code, Rain now in my city, but in windy not see the rain. Paste the old code and the new code for compare.
The first is old code :
7:12:12	Información	{"time":1743415800000,"latitude":"-38.20989","longitude":"-59.87107","temp":{"f":53.9,"c":12.17},"dewpoint":{"f":52.21,"c":11.23},"windSpeed":{"mph":6,"mps":2.68,"kph":9.66,"knots":5.21},"windGust":{"mph":13,"mps":5.81,"kph":20.92,"knots":11.3},"winddir":253,"pressure":{"inHg":29.919,"hPa":1013},"humidity":94,"windChill":{"f":51.19,"c":10.66},"heatIndex":{"f":54.16,"c":12.31},"solarRadiation":0,"precipRate":{"in":0.858,"mm":21.8},"precipSinceMidnight":{"in":null,"mm":null},"precipLast24Hours":{"in":null,"mm":null},"precipLastHour":{"in":null,"mm":null}}

7:13:44	Información	{"time":1743415800000,"latitude":"-38.20989","longitude":"-59.87107","temp":{"f":53.9,"c":12.17},"dewpoint":{"f":52.21,"c":11.23},"windSpeed":{"mph":6,"mps":2.68,"kph":9.66,"knots":5.21},"windGust":{"mph":13,"mps":5.81,"kph":20.92,"knots":11.3},"winddir":253,"pressure":{"inHg":29.919,"hPa":1013},"humidity":94,"windChill":{"f":51.19,"c":10.66},"heatIndex":{"f":54.16,"c":12.31},"solarRadiation":0,"precipRate":{"in":0,"mm":0},"precipSinceMidnight":{"in":0.079,"mm":2}}

Thanks for you help.

leoherzog, 2025-04-02:
Helpful, thank you!

Try it again. I just made more changes.

https://github.com/leoherzog/WundergroundStationForwarder/blob/davis-fix/code.gs

joseiaran, 2025-04-04:
Hi, i'm check, not rain in my city but pressure data no see in windy. The old version code see presurre. 

Thank's
Ignacio

leoherzog, 2025-04-04:
Interesting. That is unexpected. I forgot to leave commented out logging here:

https://github.com/leoherzog/WundergroundStationForwarder/blob/0bca10f50cf316ae4036174fc6441c66d835cf2a/code.gs#L421

and here:

https://github.com/leoherzog/WundergroundStationForwarder/blob/0bca10f50cf316ae4036174fc6441c66d835cf2a/code.gs#L430

Could you please add those and remove the `//` ("uncomment them") and then check the `☰▶` Executions tab in the logs?

joseiaran, 2025-04-14:
Hi:

i'm try but no change, no see presure and no see rain. The old version see presurre.

leoherzog, 2025-04-14:
In the Execution Logs, could you copy-paste the results of one of the `refreshFromDavis_()` runs after uncommenting lines 421 and 430?

leoherzog, 2025-04-15:
Aha! Davis is returning both a `bar` and a `bar_trend` and this new code was accidentally using the latter because it had "bar" in the name. I've made [another change](https://github.com/leoherzog/WundergroundStationForwarder/commit/ab201dc2f05fe3665007afcee7339474bae8d5aa) to address this, I bet this will work now. Try it one more time:

https://github.com/leoherzog/WundergroundStationForwarder/blob/davis-fix/code.gs

Thank you for your help and patience.

leoherzog, 2025-04-16:
Seems like pressure is working again!

joseiaran, 2025-04-20:
Hi:

Check the code and now no see rain in Windy. And restore code 2.9.6. The new code no see pressure and rain in windy.

leoherzog, 2025-04-20:
I made a really big architectural change a day or two ago. 

https://github.com/leoherzog/WundergroundStationForwarder/commit/7cc2e3eb7131f21cf7b94da3531253fc33ad2240

How does this most recent version work for you?

https://github.com/leoherzog/WundergroundStationForwarder/blob/davis-fix/code.gs

joseiaran, 2025-05-26:
Hi:

I'm try the code, but no rain see in Windy. 

20:54:17	Aviso	Se ha iniciado la ejecución
20:54:20	Información	{"station_id":140262,"station_id_uuid":"1a67f356-ffda-4d5c-99c6-d584aca23f94","sensors":[{"lsid":541529,"sensor_type":82,"data_structure_type":2,"data":[{"ts":1748303400,"tz_offset":-10800,"bar_trend":20,"bar":29.787,"temp_in":57.4,"hum_in":59,"temp_out":54.8,"wind_speed":19,"wind_speed_10_min_avg":18,"wind_dir":221,"temp_extra_1":null,"temp_extra_2":null,"temp_extra_3":null,"temp_extra_4":null,"temp_extra_5":null,"temp_extra_6":null,"temp_extra_7":null,"temp_soil_1":null,"temp_soil_2":null,"temp_soil_3":null,"temp_soil_4":null,"temp_leaf_1":null,"temp_leaf_2":null,"temp_leaf_3":null,"temp_leaf_4":null,"hum_out":96,"hum_extra_1":null,"hum_extra_2":null,"hum_extra_3":null,"hum_extra_4":null,"hum_extra_5":null,"hum_extra_6":null,"hum_extra_7":null,"rain_rate_clicks":4,"rain_rate_in":0.031496063,"rain_rate_mm":0.8,"uv":null,"solar_rad":0,"rain_storm_clicks":51,"rain_storm_in":0.4015748,"rain_storm_mm":10.2,"rain_storm_start_date":null,"rain_day_clicks":51,"rain_day_in":0.4015748,"rain_day_mm":10.2,"rain_month_clicks":174,"rain_month_in":1.3700787,"rain_month_mm":34.8,"rain_year_clicks":2150,"rain_year_in":16.929134,"rain_year_mm":430,"et_day":0.015,"et_month":0.183,"et_year":2.003,"moist_soil_1":null,"moist_soil_2":null,"moist_soil_3":null,"moist_soil_4":null,"wet_leaf_1":null,"wet_leaf_2":null,"wet_leaf_3":null,"wet_leaf_4":null,"forecast_rule":192,"forecast_desc":"Mostly cloudy and cooler. Precipitation possible within 12 hours, possibly heavy at times. Windy.","dew_point":53.681751854407196,"heat_index":55.175999999999995,"wind_chill":47.77012909990011,"wind_gust_10_min":27}]}],"generated_at":1748303660}
20:54:20	Información	{"time":1748303400000,"latitude":"-38.20989","longitude":"-59.87107","temp":{"f":54.8,"c":12.67},"dewpoint":{"f":53.68,"c":12.05},"windSpeed":{"mph":19,"mps":8.49,"kph":30.58,"knots":16.51},"windGust":{"mph":27,"mps":12.07,"kph":43.45,"knots":23.46},"winddir":221,"pressure":{"inHg":29.787,"hPa":1009},"humidity":96,"windChill":{"f":47.77,"c":8.76},"heatIndex":{"f":55.18,"c":12.88},"solarRadiation":0,"precipRate":{"in":0.031,"mm":0.8},"precipSinceMidnight":{"in":0.402,"mm":10.2}}
20:54:22	Información	Scheduled! Check Executions ☰▶ tab for status.
20:54:21	Aviso	Se ha completado la ejecución