leoherzog/WundergroundStationForwarder #49: help

fpcaptain, 2025-04-13:
I have a Tempest weather station.  I copied the code and changed the info to the google scripts and clicked to execute but I am not seeing anything showing that its working.  

Here are some snap shots to show that I have the info in.

[image]

[image]

[image]

I clicked the save button and Run and it says 
2:20:53 AM Notice Execution started
2:20:53 AM Notice Execution completed.

I dont see anything for KB3HLL on the Findu or APRS.fi client.  I am KB3AWQ and KB3HLL is the club I am VP of and trying to set this up.

leoherzog, 2025-04-13:
Try copying _over_ the `myFunction()` (removing everything up until the `/*` in your first line)

fpcaptain, 2025-04-13:
Syntax error: SyntaxError: Unexpected token '}' line: 1389 file: code.gs.gs 

I took out that on 1389 and then I got this 

[image]

fpcaptain, 2025-04-13:
So I am guessing this is what I should be getting

fpcaptain, 2025-04-13:
OK looks like its working now.  Thanks!  I figured it had to been something simple.

leoherzog, 2025-04-13:
Great! Glad to hear it. Consider redacting those screenshots — API keys should be treated like passwords.

closed 2025-04-13 by leoherzog