leoherzog/WundergroundStationForwarder #52: Windguru Station Type

WindySi, 2025-05-30:
Hi Leo

What station type should we be using when setting up a WindGuru station with Wunderground sstation forwarder?

We have set to other / upload API and followed the readme file and can't see any data being uploaded to our WindGuru Station

We have the Following

const datasource = 'davis'

const davisApiKey = 'API Key Number';
const davisApiSecret = 'mAPI Secret ';
const davisStationName = 'TheBeachWatersports'; 

const updateWindGuru = true;
const windGuruStationUID = 'THEBEACHWATERSPORTS';
const windGuruStationPassword = '**************';

Execution log
3:58:25 PM	Notice	Execution started
3:58:29 PM	Info	{"time":1748616300000,"latitude":"50.80234","longitude":"-0.52595"} - (This is the correct Lat/Long for the Davis Weatherlink)
3:58:32 PM	Info	Scheduled! Check Executions ☰▶ tab for status.
3:58:30 PM	Notice	Execution completed

Any ideas on why we would see problems

Many thanks for a great script and guide.

Cheers

Si

Will definitely get you some cups of tea.

WindySi, 2025-05-31:
I have noticed in the Execution updateWindGuru_ Debug	ERROR (no data)

leoherzog, 2025-06-01:
Are you trying to use a Davis weather station?

WindySi, 2025-06-01:
Yes - via the Wunderground script and Google

WindySi, 2025-06-02:
Hi Leo

I have noticed the raw data we can pull doesn't match the  Receiver requests to Davis WeatherLink

Fore instance we see the following.

           rx_state: 0,
          wind_speed_hi_last_2_min: 15.75,
          hum: 67.3,
          wind_dir_at_hi_speed_last_10_min: 225,
          wind_chill: 61.3,
          rain_rate_hi_last_15_min_clicks: 0,
          thw_index: 60.4,
          wind_dir_scalar_avg_last_10_min: 227,
          rain_size: 1,
          uv_index: null,
          tz_offset: 3600,
          wind_speed_last: 10.5,
          rainfall_last_60_min_clicks: 0,
          wet_bulb: 55.2,
          rainfall_monthly_clicks: 0,
          wind_speed_avg_last_10_min: 8.06,
          wind_dir_at_hi_speed_last_2_min: 225,
          rainfall_daily_in: 0,
          wind_dir_last: 217,
          rainfall_daily_mm: 0,
          rain_storm_last_clicks: null,
          tx_id: 1,
          rain_storm_last_start_at: null,
          rain_rate_hi_clicks: 0,
          rainfall_last_15_min_in: 0,
          rainfall_daily_clicks: 0,
          dew_point: 51.4,
          rainfall_last_15_min_mm: 0,
          rain_rate_hi_in: 0,
          rain_storm_clicks: null,
          rain_rate_hi_mm: 0,
          rainfall_year_clicks: 0,
          rain_storm_in: null,
          rain_storm_last_end_at: null,
          rain_storm_mm: null,
          wind_dir_scalar_avg_last_2_min: 223,
          heat_index: 61.5,
          rainfall_last_24_hr_in: 0,
          rainfall_last_60_min_mm: 0,
          trans_battery_flag: 0,
          rainfall_last_60_min_in: 0,
          rain_storm_start_time: null,
          rainfall_last_24_hr_mm: 0,
          rainfall_year_in: 0,
          wind_speed_hi_last_10_min: 15.75,
          rainfall_last_15_min_clicks: 0,
          rainfall_year_mm: 0,
          wind_dir_scalar_avg_last_1_min: 224,
          temp: 62.4,
          wind_speed_avg_last_2_min: 10.43,
          solar_rad: null,
          rainfall_monthly_mm: 0,
          rain_storm_last_mm: null,
          wind_speed_avg_last_1_min: 11.5,
          thsw_index: null,
          rainfall_monthly_in: 0,
          rain_rate_last_mm: 0,
          rain_rate_last_clicks: 0,
          rainfall_last_24_hr_clicks: 0,
          rain_storm_last_in: null,
          rain_rate_last_in: 0,
          rain_rate_hi_last_15_min_mm: 0,
          rain_rate_hi_last_15_min_in: 0,
          ts: 1748859345
        }
      ],
      sensor_type: 37,
      data_structure_type: 10

WindySi, 2025-06-04:
Quick update

We have managed to get the data to WindGuru by changing the sensor value and condition variable to match the above.

The only issue we see now is the time range is not picking up / sending the correct .ts and looks to be only updating .ts every 15 minutes.

we can see the following

conditions.time = davisConditions.sensors[1].data[0].ts * 1000;

Any thoughts on this

WindySi, 2025-06-05:
All fixed and working as expected - Thank you for such a great script.

closed 2025-06-05 by WindySi