leoherzog/WundergroundStationForwarder #53: Script not fetching "precipRate" and "precipSinceMidnight" data from WeatherFlow

karss79, 2025-06-20:
Hi Leo,
Over the past four days, the script has stopped sending the “precipRate” and “precipSinceMidnight” values to WU and other forwarding stations.

I checked the execution logs, and these two data points have been missing since June 16th, 2025.

I tried using different versions of your script (v2.5.0/v2.8.0/v2.9.6) nothing worked. I’m not sure if WeatherFlow has changed the way their data is being delivered.

Could you please take a look? I’ve attached screenshots of the execution logs from June 16th and 20th for reference.

Thanks

[image]
[image]

leoherzog, 2025-06-23:
Bizarre. I see that the API is returning `recip_accum_last_1hr`, `precip_accum_local_yesterday`, `precip_minutes_local_day`, and `precip_minutes_local_yesterday`, but is entirely missing `precip` and `precip_accum_local_day`. I wonder if they're rolling out a change to remove those fields? It looks like [at least one other user](https://community.tempest.earth/t/rainfall-data-not-showing-and-data-in-panel/27418) is having "missing rainfall data" issues, so perhaps consider reaching out to their support or posting on the forum?

Pseudophryne, 2025-06-23:
Yes I am experiencing the same issue, as well I faced this in Home Assistant.  The owner of the integration reached out to WeatherFlow and received this answer.

"Hi this is a known issue with our most recent API release. Hopefully sometime this week there will be a new API release and everything should start working again as normal without having to make any changes to your code."

karss79, 2025-06-24:
Hi Leo and Pseudophyrene,
Thanks for the update. I also came across another post in the WeatherFlow community forum, where a WeatherFlow staff member acknowledged the API issue. Hopefully, they’ll have it fixed soon.
https://community.tempest.earth/t/home-assistant-weatherflow-cloud-stopped-working/27426/5?u=karss
[image]

closed 2025-06-24 by leoherzog

Pseudophryne, 2025-06-28:
For anyone following this thread the issue is resolved WeatherFlow have corrected the API and all is working as expected.